programmer131/ESP8266_ESP32_SelfUpdate #1: Keep uploading firmware even if versionis same on github

happytm, 2019-12-08:
I tried you firmware but it keep uploading firmware over and over again even if version is not changed on  github. It never goes to following code in the sketch.

Serial.println("Device already on latest firmware version"); 

Serial monitor always show following message  after every message:

......Connected to WiFi
1.0

1.0
New firmware detected

Thanks.

programmer131, 2019-12-09:
Can you please share your sketch?

happytm, 2019-12-09:
It is exactly same as your Blink_ota sketch except ssid and password.

Thanks.

programmer131, 2019-12-09:
so after this message, 
"New firmware detected"
there should be LED blinking if OTA in progress, otherwise error message should appear.

happytm, 2019-12-09:
It upload new code but problem is it keep uploading even there is no change in version.

Thanks.

programmer131, 2019-12-09:
check now, issue is, you are currently using my OTA files, that i keep playing with, so maybe firmware version is different and version.txt files contains different version number, it updates firmware, then checks available version, if that is different from what is mentioned inside firmware, it restart OTA process.

happytm, 2019-12-10:
Sorry I could not make it work. But I am using this simpler code and it works. 

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define binFile "bin file location "
#define BOOT_AFTER_UPDATE    false
HTTPClient http;
 
const char* ssid = "ssid";
const char* password = "pass";

void setup()
{
  Serial.begin(115200);
  Serial.println("");
  Serial.println("Start");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.println(".");
  }                                   
  Serial.println("Connected to WiFi");
  pinMode(LED_BUILTIN, OUTPUT);
  WiFiClient client;
ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);
}

void loop()
{
   //Serial.println("This is freshly updated code from Github");
            
    ESPhttpUpdate.rebootOnUpdate(BOOT_AFTER_UPDATE);
    Serial.println("New firmware will be loaded now");
    t_httpUpdate_return ret = ESPhttpUpdate.update(binFile,"","CC AA 48 48 66 46 0E 91 53 2C 9C 7C 
    23 2A B1 74 4D 29 9D 33");
    http.end();

    switch(ret) {
         case HTTP_UPDATE_FAILED:
             Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", ESPhttpUpdate.getLastError(), 
             ESPhttpUpdate.getLastErrorString().c_str());
             break;

         case HTTP_UPDATE_NO_UPDATES:
             Serial.println("HTTP_UPDATE_NO_UPDATES");
             break;

         case HTTP_UPDATE_OK:
             Serial.println("HTTP_UPDATE_OK");
             delay(5000);                 
             ESP.restart();
             break;

         default:
             Serial.printf("Undefined HTTP_UPDATE Code: ");Serial.println(ret);
       }
  }

Thanks for your help.

closed 2019-12-10 by programmer131

programmer131, 2019-12-10:
issue is, you tried to load my binary file, latest firmware.bin file that I pushed to my repo simply puts ESP8266 in deep-sleep mode, it was for energy profiling that I demonstrated here
https://youtu.be/zbCtSRhfu5w

programmer131, 2020-12-04:
i've added payload.trim() so if there is new line character at the end of version string, it will get trimmed.