programmer131/ESP8266_ESP32_SelfUpdate #2: Error in downloading version file: -1

SudhaMaanasa, 2020-11-28:
My serial monitor shows only the following output:
Start
..............Connected to WiFi
error in downloading version file:-1
error in downloading version file:-1
error in downloading version file:-1
error in downloading version file:-1
error in downloading version file:-1

Here's my code:

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
 
const String FirmwareVer={"1.0"}; 
//#define URL_fw_Version "https://raw.githubusercontent.com/SudhaMaanasa/Smart_Device_Updates/blob/main/Version.txt"
//#define URL_fw_Bin "https://raw.githubusercontent.com/SudhaMaanasa/Smart_Device_Updates/blob/main/IPRO_4S_WifiManager_OneButton_Sinric.ino.generic.bin"
#define URL_fw_Version "https://raw.githubusercontent.com/SudhaMaanasa/Smart_Device_Updates/main/Version.txt"
#define URL_fw_Bin "https://raw.githubusercontent.com/SudhaMaanasa/Smart_Device_Updates/main/FW.bin"

//#define URL_fw_Version "http://cade-make.000webhostapp.com/version.txt"
//#define URL_fw_Bin "http://cade-make.000webhostapp.com/firmware.bin"
HTTPClient http;
 
const char* ssid = "AAA_server";
const char* password = "aaa9502547922";
 
void FirmwareUpdate()
{
   http.begin(URL_fw_Version,"5F 3F 7A C2 56 9F 50 A4 66 76 47 C6 A1 8C A0 07 AA ED BB 8E");       // check version URL
  delay(100);
  int httpCode = http.GET();            // get data from version file
  delay(100);
  String payload;
  if (httpCode == HTTP_CODE_OK)         // if version received
  {
    payload = http.getString();  // save received version
    Serial.println(payload );
  }
  else
  {
    Serial.print("error in downloading version file:");
    Serial.println(httpCode);

  }
  
  http.end();
  if (httpCode == HTTP_CODE_OK)         // if version received
  {
  payload.trim();
  if(payload.equals(FirmwareVer) )
  {   
     Serial.println("Device already on latest firmware version"); 
  }
  else
  {
     Serial.println("New firmware detected");
     WiFiClient client;

    // The line below is optional. It can be used to blink the LED on the board during flashing
    // The LED will be on during download of one buffer of data from the network. The LED will
    // be off during writing that buffer to flash
    // On a good connection the LED should flash regularly. On a bad connection the LED will be
    // on much longer than it will be off. Other pins than LED_BUILTIN may be used. The second
    // value is used to put the LED on. If the LED is on with HIGH, that value should be passed
    ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);

    t_httpUpdate_return ret = ESPhttpUpdate.update(URL_fw_Bin,"","5F 3F 7A C2 56 9F 50 A4 66 76 47 C6 A1 8C A0 07 AA ED BB 8E");
    
    switch (ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    } 
  }
 }  
}

unsigned long previousMillis = 0;        // will store last time LED was updated
const long interval = 30000;

 void repeatedCall(){
    unsigned long currentMillis = millis();
    if ((currentMillis - previousMillis) >= interval) 
    {
      // save the last time you blinked the LED
      previousMillis = currentMillis;
      FirmwareUpdate();
    }
 }

  
void setup()
{
  Serial.begin(115200);
  Serial.println("");
  Serial.println("Start");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print("..");
  }                                   
  Serial.println("Connected to WiFi");
  pinMode(LED_BUILTIN, OUTPUT);
}
void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
  repeatedCall();    
}

Could you please help me with the correction of my code?

programmer131, 2020-11-29:
pull the latest version, i just updated fingerprint for raw.githubusercontent.com, need to add a way to auto update fingerprint after fw is deployed.

closed 2020-11-29 by programmer131

SudhaMaanasa, 2020-11-29:
Can you please specify the steps to collect the fingerprints?

programmer131, 2020-11-29:
https://www.grc.com/fingerprints.htm

SudhaMaanasa, 2020-11-29:
Thank you.. could you please explain to me the meaning of "pulling the latest version"

programmer131, 2020-11-29:
download updated code from github,

SudhaMaanasa, 2020-11-29:
Ok Thank you.. I tried the code that u had mentioned. But no improvement. the same outcome.

SudhaMaanasa, 2020-11-29:
The code you mentioned as an example is working but when I m replacing it with my code it isn't working.

Here's my code:

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
 
const String FirmwareVer={"1.1"}; 
#define URL_fw_Version "https://raw.githubusercontent.com/SudhaMaanasa/Smart_Updates/master/Version.txt"
#define URL_fw_Bin "https://raw.githubusercontent.com/SudhaMaanasa/Smart_Updates/master/firmware.bin"

//#define URL_fw_Version "http://cade-make.000webhostapp.com/version.txt"
//#define URL_fw_Bin "http://cade-make.000webhostapp.com/firmware.bin"
HTTPClient http;
 
const char* ssid = "Your Wifi SSID";
const char* password = "Your Wifi Pswd";
 
void FirmwareUpdate()
{  
  http.begin(URL_fw_Version,"5F 3F 7A C2 56 9F 50 A4 66 76 47 C6 A1 8C A0 07 AA ED BB 8E");     // check version URL
  int httpCode = http.GET();            // get data from version file
  String payload;
  if (httpCode == HTTP_CODE_OK)         // if version received
  {
    payload = http.getString();  // save received version
    Serial.println(payload );
  }
  else
  {
    Serial.print("error in downloading version file:");
    Serial.println(httpCode);
  }
  
  http.end();
  if (httpCode == HTTP_CODE_OK)         // if version received
  {
  payload.trim();
  if(payload.equals(FirmwareVer) )
  {   
     Serial.println("Device already on latest firmware version"); 
  }
  else
  {
     Serial.println("New firmware detected");
     WiFiClient client;

    // The line below is optional. It can be used to blink the LED on the board during flashing
    // The LED will be on during download of one buffer of data from the network. The LED will
    // be off during writing that buffer to flash
    // On a good connection the LED should flash regularly. On a bad connection the LED will be
    // on much longer than it will be off. Other pins than LED_BUILTIN may be used. The second
    // value is used to put the LED on. If the LED is on with HIGH, that value should be passed
    ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);

    t_httpUpdate_return ret = ESPhttpUpdate.update(URL_fw_Bin,"","5F 3F 7A C2 56 9F 50 A4 66 76 47 C6 A1 8C A0 07 AA ED BB 8E");
    
    switch (ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    } 
  }
 }  
}

unsigned long previousMillis = 0;        // will store last time LED was updated
const long interval = 10000;

 void repeatedCall(){
    unsigned long currentMillis = millis();
    if ((currentMillis - previousMillis) >= interval) 
    {
      // save the last time you blinked the LED
      previousMillis = currentMillis;
      FirmwareUpdate();
    }
 }

  
void setup()
{
  Serial.begin(115200);
  Serial.println("");
  Serial.println("Start");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }                                   
  Serial.println("Connected to WiFi");
  pinMode(LED_BUILTIN, OUTPUT);
}
void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
  repeatedCall();    
}

SudhaMaanasa, 2020-11-29:
Yes thq it worked.

programmer131, 2020-12-04:
Hi, check latest version, i've removed fingerprints from code.