programmer131/ESP8266_ESP32_SelfUpdate #3: DigiCert High Assurance EV Root CA Generation

Maxvi33, 2020-12-08:
Hello,
this certificate is specific to you ? It's possible to explain how generate own proper certificate for ESP ?

programmer131, 2020-12-08:
no it isn't specific to me, i'm not using my own server, it is for github. if you have your own domain where you want to host binary file, then you can use your own self signed certificates.

emailsidbhatia, 2020-12-15:
Hi @programmer131! How did you generate the certificate for github and what is the use of the setClock() function?
Thanks

programmer131, 2021-01-21:
We can download certificate from Firefox, I didn't generate any certificate myself.
setClock() function takes time from NTP server and configure it into esp8266, it is used for x.509 validation.

closed 2021-02-15 by programmer131