programmer131/ESP8266_ESP32_SelfUpdate #4: A Request

malosaa, 2020-12-08:
Hi, 

wow this code works really well. 
i just tested it and did it awsome.

is there a way to accept before updating directly.
Like a button on a html page that will let u update instead of forcing the update directly

regards

programmer131, 2020-12-08:
That should be easy to do, are you referring to esp32 or esp8266 ota code? Button press on esp32 can start OTA, same can be triggered from mqtt for example, send an mqtt message that performs OTA.

malosaa, 2020-12-08:
@programmer131 

yeah for esp8266, i would like to add a html button to trigger the new update.

but i have encountered a little problem, because of the certificate when accessing my webpage on my arduino crashes it.. 
i don't know why. when i remove the certificate from the code it doesnt crash anymore,  but ofcourse the auto update wont work anymore.

regards

malosaa, 2020-12-08:
this is the error i get when i dont access the webserver after a while

if i access it while it loads up i get this error directly

```
Idle Loop....178
Waiting for NTP time sync: 
Abort called

>>>stack>>>

ctx: cont
sp: 3fff0fc0 end: 3fff1390 offset: 01b0
3fff1170:  c02ac026 c00ec004 3fff11a0 40208ad7  
3fff1180:  c09dc09c c0a1c0a0 003d003c 0035002f  
3fff1190:  00000000 3fff73fc 3fff1240 40208d4f  
3fff11a0:  00000000 00000000 3fff11d4 00000000  
3fff11b0:  3ffe8ed0 3fff73fc 3fff1240 40209674  
3fff11c0:  00000000 3fff1220 402197f8 3fff0370  
3fff11d0:  00000000 00000000 00000000 40219ff7  
3fff11e0:  00000000 3fff4e84 3fff1240 40208309  
3fff11f0:  000001bb 00000d50 3fff12bc 3ffefd94  
3fff1200:  000001bb 3fff1240 3ffe8ed0 3ffefd94  
3fff1210:  000001bb 3fff1240 3ffe8ed0 40209865  
3fff1220:  40223d58 85006597 40223d58 85006597  
3fff1230:  3fffdad0 3ffe8544 001b8464 40204f52  
3fff1240:  40223ad8 00000000 00003a98 00000018  
3fff1250:  3ffef810 3fff4e84 3fff66a4 3fff25a4  
3fff1260:  00000000 00000000 3fff66a4 00000000  
3fff1270:  00000000 00000000 00000000 00000000  
3fff1280:  00000000 3fff73fc 3fff244c 3fff3bec  
3fff1290:  3fff3e4c 00000000 3ffef9b0 00000000  
3fff12a0:  00004145 00000255 3fff0000 00000000  
3fff12b0:  00000000 00000000 00000000 00000000  
3fff12c0:  00000000 00000000 3ffe0000 4021b8de  
3fff12d0:  00000000 4bc6a7f0 3fff0338 3f00fd94  
3fff12e0:  00000000 00000000 00000000 00000000  
3fff12f0:  3fff1500 00000000 00000000 40217019  
3fff1300:  00000000 00000000 3ffef5d4 4020e854  
3fff1310:  1fd11445 3ffefba4 3ffefc50 40217204  
3fff1320:  3fffdad0 00000000 3ffefc50 40201f3a  
3fff1330:  3fffdad0 00000000 001b8464 4020512b  
3fff1340:  3fffdad0 00000000 3ffef5d4 40205bf1  
3fff1350:  00000000 feefeffe feefeffe feefeffe  
3fff1360:  00000000 00000000 00000001 3ffefd94  
3fff1370:  3fffdad0 00000000 3ffefd64 402198a8  
3fff1380:  feefeffe feefeffe 3ffe8588 4010065d  
<<<stack<<<

last failed alloc call: 4022CEFE(3064)
```

malosaa, 2020-12-10:
i did a debug and did see this, 

Waiting for NTP time sync: BSSL:_connectSSL: start connection
BSSL:Connected!
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
BSSL:read: failed
Device already on latest firmware version

i did shorten the output, but is there any way to stop this multi data? 
i think this is causing the error.

my arduino has a webserver on it with jquery and ajax, i think when i access the website while its checking for an update of firmware it crashes the arduino..

malosaa, 2020-12-10:
also @programmer131 can u pprovide me an example on esp8266 that it only checkes for an update when a button on a webpage is pressed? instead of checking every x seconds.
i think thats causes my crash, because im running a webserver on the arduino with ajax and jquery on xhttp

regards and thanks for this awsome release !

closed 2021-02-15 by programmer131