programmer131/ESP8266_ESP32_SelfUpdate #6: reboot

Maxvi33, 2021-01-14:
Hello i try to use your soft on new esp8266 but this is behavior:

> Active fw version:1.1
> Idle Loop....33
> Waiting for NTP time sync: 
> Current time: Thu Jan 14 15:23:12 2021
> New firmware detected
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 3584, room 16 
> tail 0
> chksum 0xb0
> csum 0xb0
> v2843a5ac
> @cp:0
> ld
> 
> Start
> OOOOOOOOOOOConnected to WiFi with soft 2
> Waiting for NTP time sync: ...............................
> Current time: Thu Jan 14 15:23:59 2021
>  Active fw version:1.1
> Idle Loop....0
>  Active fw version:1.1
> Idle Loop....1
>  Active fw version:1.1

do you have an idea why he reboot ? ",rst cause:2, boot mode:(3,6)"

fridayowl, 2021-01-15:
am getting this same issue.

star297, 2021-01-15:
Works on ESP8266 and you should end up with the following terminal message:
(don't forget the system should reboot as we've just overwritten the original program) 

 
Idle Loop....51
 Active fw version:1.8
Idle Loop....52
Waiting for NTP time sync: New firmware detected

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
@cp:0
ld

device just uploaded new firmware....
current firmware version is 1.9
you won't be able to connect to WiFi, do you know why?

Nice job @programmer131

programmer131, 2021-01-21:
reboot happens after new firmware is detected and downloaded, another issue maybe, you have updated firmware version string inside program but didn't update version string in text file that ESP8266 is polling, so it thinks that there is new firmware out there. it will keep uploading new binary every time.

closed 2021-02-15 by programmer131