programmer131/ESP8266_ESP32_SelfUpdate #7: Firebase Storage OTA download for ESP8266 ?

star297, 2021-01-15:
I can use fota my ESP32's using Firebase Storage.
See here:

[https://medium.com/firebase-developers/over-the-air-arduino-firmware-updates-using-firebase-part-1-3e874ad9f93e](url)

This works flawless on the ESP32, but no go on the ESP8266 

Your example works for me, however I need to change the URL_fw_Bin to something like this:

https://firebasestorage.googleapis.com/v0/b/ProjectName.appspot.com/o/ESP8266-BME280-moisture.ino.generic.bin?alt=media&token=7b16109d-dc69-491b-baf1-6c50c4e3b247

Is this possible or just stick to the ESP32 ?

I don't have a "host" as such so can't use `client.connect(host, httpsPort)` 

I'm using ` http.begin(FIRMWARE_URL);` on the ESP32.

programmer131, 2021-01-15:
if you can share some working firebase url, i can test here. just upload a dummy binary and share url.

star297, 2021-01-15:
Hi Farrukh,

Here is the URL to a random ESP8266 program on my Firebase Storage test site.

[https://firebasestorage.googleapis.com/v0/b/mbed-a5059.appspot.com/o/ESP8266-BME280-moisture.ino.generic.bin?alt=media&token=454bad1d-562d-41f3-9e17-e7df4447226d](url)

It would be nice if you can work you code round to be able to download and update like I am with the ESP32.

Thanks, Paul

programmer131, 2021-01-21:
#define URL_fw_Bin "https://firebasestorage.googleapis.com/v0/b/mbed-a5059.appspot.com/o/ESP8266-BME280-moisture.ino.generic.bin?alt=media&token=454bad1d-562d-41f3-9e17-e7df4447226d"
update root certificate and it should work

star297, 2021-01-22:
Thanks Farrukh,
The basic code below works every time with this certificate I use with Mbed and is simple to integrate with existing programs.
If Firebase code is running you must stop it by calling something like `Firebase.end(firebaseData);` before running the update code otherwise it will crash.
Best to use ESP devices with at least 4Mbyte flash.
The host is the Firebase project's databaseURL.

```
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <WiFiClientSecure.h>
#include <CertStoreBearSSL.h>
BearSSL::CertStore certStore;

#define URL_fw_Bin "https://firebasestorage.googleapis.com/v0/b/.....  The FIREBASE STORAGE uploaded file link here"
const char* host = "https://project_name.firebaseio.com";
const int httpsPort = 443;

const char* ssid = "ssid";
const char* password = "pw";

const char trustRoot[] =
  "-----BEGIN CERTIFICATE-----\n"
  "MIIESjCCAzKgAwIBAgINAeO0mqGNiqmBJWlQuDANBgkqhkiG9w0BAQsFADBMMSAw\n"
  "HgYDVQQLExdHbG9iYWxTaWduIFJvb3QgQ0EgLSBSMjETMBEGA1UEChMKR2xvYmFs\n"
  "U2lnbjETMBEGA1UEAxMKR2xvYmFsU2lnbjAeFw0xNzA2MTUwMDAwNDJaFw0yMTEy\n"
  "MTUwMDAwNDJaMEIxCzAJBgNVBAYTAlVTMR4wHAYDVQQKExVHb29nbGUgVHJ1c3Qg\n"
  "U2VydmljZXMxEzARBgNVBAMTCkdUUyBDQSAxTzEwggEiMA0GCSqGSIb3DQEBAQUA\n"
  "A4IBDwAwggEKAoIBAQDQGM9F1IvN05zkQO9+tN1pIRvJzzyOTHW5DzEZhD2ePCnv\n"
  "UA0Qk28FgICfKqC9EksC4T2fWBYk/jCfC3R3VZMdS/dN4ZKCEPZRrAzDsiKUDzRr\n"
  "mBBJ5wudgzndIMYcLe/RGGFl5yODIKgjEv/SJH/UL+dEaltN11BmsK+eQmMF++Ac\n"
  "xGNhr59qM/9il71I2dN8FGfcddwuaej4bXhp0LcQBbjxMcI7JP0aM3T4I+DsaxmK\n"
  "FsbjzaTNC9uzpFlgOIg7rR25xoynUxv8vNmkq7zdPGHXkxWY7oG9j+JkRyBABk7X\n"
  "rJfoucBZEqFJJSPk7XA0LKW0Y3z5oz2D0c1tJKwHAgMBAAGjggEzMIIBLzAOBgNV\n"
  "HQ8BAf8EBAMCAYYwHQYDVR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMBIGA1Ud\n"
  "EwEB/wQIMAYBAf8CAQAwHQYDVR0OBBYEFJjR+G4Q68+b7GCfGJAboOt9Cf0rMB8G\n"
  "A1UdIwQYMBaAFJviB1dnHB7AagbeWbSaLd/cGYYuMDUGCCsGAQUFBwEBBCkwJzAl\n"
  "BggrBgEFBQcwAYYZaHR0cDovL29jc3AucGtpLmdvb2cvZ3NyMjAyBgNVHR8EKzAp\n"
  "MCegJaAjhiFodHRwOi8vY3JsLnBraS5nb29nL2dzcjIvZ3NyMi5jcmwwPwYDVR0g\n"
  "BDgwNjA0BgZngQwBAgIwKjAoBggrBgEFBQcCARYcaHR0cHM6Ly9wa2kuZ29vZy9y\n"
  "ZXBvc2l0b3J5LzANBgkqhkiG9w0BAQsFAAOCAQEAGoA+Nnn78y6pRjd9XlQWNa7H\n"
  "TgiZ/r3RNGkmUmYHPQq6Scti9PEajvwRT2iWTHQr02fesqOqBY2ETUwgZQ+lltoN\n"
  "FvhsO9tvBCOIazpswWC9aJ9xju4tWDQH8NVU6YZZ/XteDSGU9YzJqPjY8q3MDxrz\n"
  "mqepBCf5o8mw/wJ4a2G6xzUr6Fb6T8McDO22PLRL6u3M4Tzs3A2M1j6bykJYi8wW\n"
  "IRdAvKLWZu/axBVbzYmqmwkm5zLSDW5nIAJbELCQCZwMH56t2Dvqofxs6BBcCFIZ\n"
  "USpxu6x6td0V7SvJCCosirSmIatj/9dSSVDQibet8q/7UK4v4ZUN80atnZz1yg==\n"
  "-----END CERTIFICATE-----\n";

X509List cert(trustRoot);

void FirmwareUpdate()
{
  WiFiClientSecure client;
  client.setTrustAnchors(&cert);
  if (!client.connect(host, httpsPort)) {
    Serial.println("Connection failed");
    return;
  }
  Serial.println("Updateing");
  ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);
  t_httpUpdate_return ret = ESPhttpUpdate.update(client, URL_fw_Bin);
  switch (ret) {
    case HTTP_UPDATE_FAILED:
      Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
      break;
    case HTTP_UPDATE_NO_UPDATES:
      Serial.println("HTTP_UPDATE_NO_UPDATES");
      break;
      // dos'nt actully get here, the ESP restarts with the downloaded program BIN file 
    case HTTP_UPDATE_OK:
      Serial.println("HTTP_UPDATE_OK");
      break;
  }
}

void setup()
{
  Serial.begin(115200);
  Serial.println("");
  Serial.println("Start");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print("O");
  }
  Serial.println("Connected to WiFi");
  const char * NTPpool         = "nl.pool.ntp.org";
  const char * defaultTimezone = " CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00";
  configTzTime( defaultTimezone, NTPpool); //sets TZ and starts NTP sync
  struct tm timeinfo = {0};

  // give some time here to get NTP
  int timeout = 0;
  while (time(nullptr) < 1611253722 && timeout <= 20) {
    timeout++;
    delay(100);
  }

  pinMode(LED_BUILTIN, OUTPUT);
  FirmwareUpdate();

}
void loop()
{
}
```

closed 2021-02-15 by programmer131

brama-coder, 2021-11-08:
I am trying to perform the same task as you @star297, however when running the above code after changing network credentials and urls I am not getting any serial monitor output?

star297, 2021-11-08:
You could try this:
https://esp-updater.netlify.app/

brama-coder, 2021-11-09:
This is great! However, I would like to be able to upload my firmware file
in one location where my other esp devices could pull from. In your code
above, where do I retrieve the certificate from? What is its purpose? Could
that be the reason why I am getting a connection failure?