programmer131/ESP8266_ESP32_SelfUpdate #8: Error 101 Server Did Not Report Size from OTA Upgrade framework via git hub

soheilpaper, 2021-01-21:
I have tried [this video][1]:

[ESP8266 Self Update OTA from GitHub][1]

Instruction and made [this GitHub repo][2]:

And changed [the main code][3] to [this type][4], but I get this error:

```
HTTP_UPDATE_FAILD Error (-101): Server Did Not Report Size
```
[![enter image description here][5]][5]
 when the `   FirmwareUpdate();` code is running. 
So I don't know why GitHub don't answer to the esp8266, as said [here][6]:

> The error has nothing to do with your sketch. Also, the error code is
> not a HTTP 101 error. The problem is that your server does not send
> the response header 'Content-Length'

I have some dupt about the bin URL file address which have many dot on the end of it (`https://github.com/soheilpaper/ota_github_blink/blob/main/BlinkWithoutDelay.ino.esp8285.bin`) on it, could it be the reason for none answering and giving the 101 code form GitHub server!

Any suggestion would be good help to fix this bug.

**Update:**

> I have changed the bin file name to this kind :
> 
>      BlinkWithoutDelay.ino.esp8285.bin  ---->      BlinkWithoutDelay_ino_esp8285.bin
> 
>  But I have again the same GitHub 101 error.

 Also asked here:
https://arduino.stackexchange.com/questions/80890/error-101-server-did-not-report-size-from-ota-upgrade-framework-via-git-hub

Thanks.

  [1]: https://www.youtube.com/watch?v=FuY6BobS-1k
  [2]: https://github.com/soheilpaper/ota_github_blink
  [3]: https://github.com/programmer131/ESP8266_ESP32_SelfUpdate/blob/master/esp8266_ota/esp8266_ota.ino
  [4]: https://github.com/soheilpaper/ota_github_blink/blob/main/ESP_OTA_Github.ino
  [5]: https://i.stack.imgur.com/ZI74L.png
  [6]: https://github.com/esp8266/Arduino/issues/5131

programmer131, 2021-01-21:
there is issue with your firmware url, it should be something like this
URL_fw_Bin "https://raw.githubusercontent.com/programmer131/otaFiles/master/firmware.bin"
your link points to webpage, not to bin file
https://github.com/soheilpaper/ota_github_blink/blob/main/BlinkWithoutDelay_ino_esp8285.bin

soheilpaper, 2021-01-21:
Thanks  
Based on the [timemage][7] answer via [the other question](https://arduino.stackexchange.com/questions/80890/error-101-server-did-not-report-size-from-ota-upgrade-framework-via-git-hub?noredirect=1#comment180927_80890), I have changed the address of `bin` and `TXT` file by finding the download link of those files form the GitHub. And it has been working by changing the `blob` part of address to `raw`, so the bin address changed form `https://github.com/soheilpaper/ota_github_blink/blob/main/BlinkWithoutDelay_ino_esp8285.bin` to `https://github.com/soheilpaper/ota_github_blink/raw/main/BlinkWithoutDelay_ino_esp8285.bin` And the `https://github.com/soheilpaper/ota_github_blink/blob/main/version.txt`to  `https://raw.githubusercontent.com/soheilpaper/ota_github_blink/main/version.txt`

And now that error has been fixed as you can see below:

[![enter image description here][8]][8]

Although I gave some dupt about upgrading the main code by getting this output via serial port:

```
Device already on latest firmware version

```
And I would check it after some resting!.

Thanks.

  [1]: https://www.youtube.com/watch?v=FuY6BobS-1k
  [2]: https://github.com/soheilpaper/ota_github_blink
  [3]: https://github.com/programmer131/ESP8266_ESP32_SelfUpdate/blob/master/esp8266_ota/esp8266_ota.ino
  [4]: https://github.com/soheilpaper/ota_github_blink/blob/main/ESP_OTA_Github.ino
  [5]: https://i.stack.imgur.com/ZI74L.png
  [6]: https://github.com/esp8266/Arduino/issues/5131
  [7]: https://arduino.stackexchange.com/users/70020/timemage
  [8]: https://i.stack.imgur.com/z6cw8.png

soheilpaper, 2021-01-21:
> there is issue with your firmware url, it should be something like this
> URL_fw_Bin "https://raw.githubusercontent.com/programmer131/otaFiles/master/firmware.bin"
> your link points to webpage, not to bin file
> https://github.com/soheilpaper/ota_github_blink/blob/main/BlinkWithoutDelay_ino_esp8285.bin

Thanks  
Based on the [timemage][7] answer via [the other question](https://arduino.stackexchange.com/questions/80890/error-101-server-did-not-report-size-from-ota-upgrade-framework-via-git-hub?noredirect=1#comment180927_80890), I have changed the address of `bin` and `TXT` file by finding the download link of those files form the GitHub. And it has been working by changing the `blob` part of address to `raw`, so the bin address changed form `https://github.com/soheilpaper/ota_github_blink/blob/main/BlinkWithoutDelay_ino_esp8285.bin` to `https://github.com/soheilpaper/ota_github_blink/raw/main/BlinkWithoutDelay_ino_esp8285.bin` And the `https://github.com/soheilpaper/ota_github_blink/blob/main/version.txt`to  `https://raw.githubusercontent.com/soheilpaper/ota_github_blink/main/version.txt`

And now that error has been fixed as you can see below:

[![enter image description here][8]][8]

Although I gave some dupt about upgrading the main code by getting this output via serial port:

```
Device already on latest firmware version

```
And I would check it after some resting!.

Thanks.

  [1]: https://www.youtube.com/watch?v=FuY6BobS-1k
  [2]: https://github.com/soheilpaper/ota_github_blink
  [3]: https://github.com/programmer131/ESP8266_ESP32_SelfUpdate/blob/master/esp8266_ota/esp8266_ota.ino
  [4]: https://github.com/soheilpaper/ota_github_blink/blob/main/ESP_OTA_Github.ino
  [5]: https://i.stack.imgur.com/ZI74L.png
  [6]: https://github.com/esp8266/Arduino/issues/5131
  [7]: https://arduino.stackexchange.com/users/70020/timemage
  [8]: https://i.stack.imgur.com/z6cw8.png

programmer131, 2021-02-15:
closing as issue is already fixed.

closed 2021-02-15 by programmer131