programmer131/ESP8266_ESP32_SelfUpdate #13: ESP32_SelUpdate Error

HHavendano, 2022-07-04:
Hi, i tried to use the ESP32's code ,and it gave me error -1, i changed the fingerprints, but it's the same error. Do you know what is my error? i tried this code ,and another i found over there. And it's the same error.

allanaus, 2022-07-05:
The cert.h file is outdated.  Verbose output in the serial monitor will show you this error.
[ 60859][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():271]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed.
Try using the updated certificate from here:
https://github.com/allanaus/ESP8266_ESP32_SelfUpdate/blob/master/cert.h
Just checked it, the code works fine.

HHavendano, 2022-07-11:
Hi, thank you so much, if i want to update the cert.h in 2031, where can i found it?

programmer131, 2022-07-14:
Thanks @allanaus 
I've updated cert for both esp8266 and esp32, 
@HHavendano One simple option might be to use BearSSL and insecure client, it will allow communication over TLS/HTTPS, this way you can totally remove certificates.
see the get_sunrise_sunset_time() function in this sketch, where I've used same approach, [cam light project](https://github.com/programmer131/cam-flood-light/blob/main/cam-v3.ino)

HHavendano, 2022-07-15:
Thanks @programmer131, I´ll use it for the next proyect, but i would want to know where to find the cert.h

allanaus, 2022-07-15:
To download a certificate for any site, in Chrome go to its webpage, in this case it is "https://raw.githubusercontent.com/programmer131/ESP8266_ESP32_SelfUpdate/master/esp32_ota/bin_version.txt".
Click on the padlock in the address bar -> then "Connection is secure" -> then "Certificate is valid".
In the details tab click on "Copy to file" -> then "Next".  ESP32 uses the "Base-64 encoded X.509" format, select that.
Give the file a name, then download your certificate.  This is just one way of doing it, but I find it the easiest.
ESP8266 uses the DER format that will need to be converted so the process is a little different.

HHavendano, 2022-07-15:
@allanaus ok, i downloaded the certificate, where is the code to add it?

BlaineAtkins, 2022-07-27:
> Thanks @allanaus I've updated cert for both esp8266 and esp32, @HHavendano One simple option might be to use BearSSL and insecure client, it will allow communication over TLS/HTTPS, this way you can [...]

@programmer131 Thank you so much for this example (and your library as well, I love it!). I would love to get rid of CA certificates entierly, but the one problem is I am using ESP32, and BearSSL does not support ESP32 (only ESP8266). Have you been able to find some way to eliminate need for CA certificates on ESP32 platform? Thank you.

BlaineAtkins, 2022-07-27:
Oops, sorry, nevermind. I just discovered that the setInsecure() method works without BearSSL, and also works in the standard WiFiClientSecure library. I am very excited to not have to manually update certificates again in 10 years haha.

ngocthuong234, 2022-08-06:
> get_sunrise_sunset_time

sorry my english is very bad. Can you give more detailed instructions?

ngochan2112, 2022-09-01:
> Oops, sorry, nevermind. I just discovered that the setInsecure() method works without BearSSL, and also works in the standard WiFiClientSecure library. I am very excited to not have to manually update certificates again in 10 years haha.

Can you give more detailed instructions? I am a newbie

BlaineAtkins, 2022-09-02:
> > Oops, sorry, nevermind. I just discovered that the setInsecure() method works without BearSSL, and also works in the standard WiFiClientSecure library. I am very excited to not have to manually update certificates again in 10 years haha.
> 
> Can you give more detailed instructions? I am a newbie

If you have a WiFiClient like:
`WiFiClientSecure client;`
you would usually set a CA certificate for authentication like:
`client.setCACert(rootCACertificate);`.
But if you want to bypass authentication and not use a certificate, instead of the above line, you can put:
`client.setInsecure();`.
Now you can communicate with the server without a cert.

ngocthuong234, 2022-09-06:
> If you have a WiFiClient like: `WiFiClientSecure client;` you would usually set a CA certificate for authentication like: `client.setCACert(rootCACertificate);`. But if you want to bypass [...]

Thank you very much, it worked!

HHavendano, 2022-12-08:
I have a problem, my co-worker created a website that i need compulsorily add cert in each module. The problem is the modules will be distributed for all the state. So i have the way to update the cert automatically every 10 years remotely. Any ideas?